projectchrono/chrono #452: cannot find demo_FSI_Flexible_Shell.cpp

alundilong, 2023-05-29:
As shown in the doc, there is a FSI example, but I cannot find any. Please help.
[image]

rserban, 2023-05-29:
The Chrono::FSI demos are all under src/demos/fsi/:
https://github.com/projectchrono/chrono/tree/main/src/demos/fsi

Are you by any chance using PyChrono? If so, note that the FSI module is currently not wrapped for use with Python.

alundilong, 2023-05-31:
Hi 

> The Chrono::FSI demos are all under src/demos/fsi/: https://github.com/projectchrono/chrono/tree/main/src/demos/fsi
> 
> Are you by any chance using PyChrono? If so, note that the FSI module is currently not wrapped for use with Python.

Thank you for the answer. I am using C++ version of Chrono. I guess it was sort of typo, as I did not see any file "demo_FSI_Flexible_Shell.cpp" either at https://github.com/projectchrono/chrono/tree/main/src/demos/fsi
[image]

rserban, 2023-05-31:
Ah, I understand now what the issue was. There were some changes in demo names and we forgot to update the documentation.  I made a not to fix that. Thanks for pointing that out.