projectchrono/chrono #460: The ARM64 development board performs a make check error

xcmblueskys, 2023-07-22:
29/49 Testing: utest_FEA_ANCFCable
29/49 Test: utest_FEA_ANCFCable
Command: "/home/xucaiming/workspaces/master/tpc_c_cplusplus/thirdparty/chrono/chrono-8.0.0/arm64-v8a-build/bin/utest_FEA_ANCFCable"
Directory: /home/xucaiming/workspaces/master/tpc_c_cplusplus/thirdparty/chrono/chrono-8.0.0/arm64-v8a-build/bin
"utest_FEA_ANCFCable" start time: Aug 05 09:04 UTC
Output:
----------------------------------------------------------
Time t = 0.0001s
Time t = 0.0002s
Time t = 0.0003s
Time t = 0.0004s
Time t = 0.0005s
Time t = 0.0006s
Time t = 0.0007s
Time t = 0.0008s
Time t = 0.0009s
Time t = 0.001s
Time t = 0.0011s
Time t = 0.0012s
Time t = 0.0013s
Time t = 0.0014s
Time t = 0.0015s
Time t = 0.0016s
Unit test check failed
  y position: -1.19492e-05  (reference: -1.3061e-05  diff: 1.11175e-06)
  z position: -0.00640117  (reference: -0.00640121  diff: 3.79896e-08)
<end of output>
Test time =   1.34 sec
----------------------------------------------------------
Test Failed.
"utest_FEA_ANCFCable" end time: Aug 05 09:04 UTC
"utest_FEA_ANCFCable" time elapsed: 00:00:01
----------------------------------------------------------

Looking forward to hearing from you

xcmblueskys, 2023-07-22:
Looking forward to hearing from you

uebian, 2026-01-12:
@xcmblueskys I'm able to pass the test `utest_FEA_ANCFCable` on my arm64 device. Could you please share the specific model of your device and the distribution you are using?