projectchrono/chrono #469: AttributeError:'ChSystemNSC' object has no attribute 'SetVisualSystem'

EQJ1NBJ, 2023-10-12:
version: Solidworks2021 chrono 8.0 and Chronosolidworks8.0
question: when do the solidworks demo engine，after click the 'Run test', there is the Error hit:
AttributeError:'ChSystemNSC' object has no attribute 'SetVisualSystem'
[image]

rserban, 2023-10-12:
Could you please confirm that replacing line 198 in `run_test.py` with
`    vis.AttachSystem(my_system)`
fixes the problem?

EQJ1NBJ, 2023-10-12:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        Hello Serban,
        Thanks for the reply.I have tried as your suggestion, then there appears a new similar error.Wang Nan
    
        
            
        
    
    
     
    
        
                    ***@***.***
        
    
 

  
    ---- Replied Message ----
  
  
    
      
        
         From 
        
        
            Radu ***@***.***>
            
        
      
      
        
         Date 
        
        
        10/12/2023 13:56
        
      
      
        
         To 
        
        
         
          
            ***@***.***>
            
          
        
      
      
        
         Cc 
        
        
          
            ***@***.***>
            ,
          
          
            ***@***.***>
            
          
        
      
      
        
         Subject 
        
        
              Re: [projectchrono/chrono] AttributeError:'ChSystemNSC' object has no attribute 'SetVisualSystem' (Issue #469)
        
      
    
  
  
Could you please confirm that replacing line 198 in run_test.py with
    vis.AttachSystem(my_system)
fixes the problem?

rserban, 2023-10-12:
I'm not sure I follow.  Could you please be more clear in describing whatever issue it is that you are having?

EQJ1NBJ, 2023-10-12:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        
            
        
    
    
        sorry for missing the attachment.
        
    
     
    
        
                    ***@***.***
        
    
 

  
    ---- Replied Message ----
  
  
    
      
        
         From 
        
        
            Radu ***@***.***>
            
        
      
      
        
         Date 
        
        
        10/12/2023 15:17
        
      
      
        
         To 
        
        
         
          
            ***@***.***>
            
          
        
      
      
        
         Cc 
        
        
          
            ***@***.***>
            ,
          
          
            ***@***.***>
            
          
        
      
      
        
         Subject 
        
        
              Re: [projectchrono/chrono] AttributeError:'ChSystemNSC' object has no attribute 'SetVisualSystem' (Issue #469)
        
      
    
  
  
I'm not sure I follow.  Could you please be more clear in describing whatever issue it is that you are having?

rserban, 2023-10-12:
I still don't understand.  What is the problem? What error do you get? Doing what?
What is this snippet:
```
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }
```
supposed to say?

rserban, 2023-10-12:
@dariofusai93 Do you have any idea what this could be about?!?

EQJ1NBJ, 2023-10-12:
> @dariofusai93 Do you have any idea what this could be about?!?

I replyed  you with the error picture just now by E-mail.
I just find that you can't see the error picture.
now I put the error picture again here.
[image]

rserban, 2023-10-12:
OK, that Python script must be fixed (for both errors). We'll do that and will post here when done.

In the future, please just type the error message instead of trying to attach images.

rserban, 2023-10-12:
There's an updated SolidWorks plugin on the download page on [projectchrono.org](https://projectchrono.org/download/).
Please try installing the new plugin and let us know if this resolves the issue.

dariomangoni, 2023-10-12:
@EQJ1NBJ please mind that the Solidworks plugin is now fixed and is expected to work fine, but with the **current development version** of Chrono so it is not guaranteed that will work when exporting to the older official binary release of Chrono **8.0**!

EQJ1NBJ, 2023-10-13:
> There's an updated SolidWorks plugin on the download page on [projectchrono.org](https://projectchrono.org/download/). Please try installing the new plugin and let us know if this resolves the issue.

I have tried to download the Solidworks9.2 on this page mane times, but failed. other version can still be download.

rserban, 2023-10-13:
I just reorganized how access to the SolidWorks add-in versions is provided.  Please see the updated [download](https://projectchrono.org/download/) page on the project website. 
- a direct link is provided to download the version of the add-in that is compatible with the **current** Chrono development code.
- versions of the add-in compatible with previous Chrono releases are now available from the GitHub Releases page.

Please try to get the new SolidWorks add-in (now called `Chrono_SolidWorks_current.exe`) and let us know if everything works.

EQJ1NBJ, 2023-10-15:
> I just reorganized how access to the SolidWorks add-in versions is provided. Please see the updated [download](https://projectchrono.org/download/) page on the project website.
>
> * a direct link is [...]

Thanks.
I have tried the current Chronosolidworks  with both Chrono 8.0 and development branch, the 8.0 is not fixed, but the development branch is OK.

rserban, 2023-10-16:
As I mentioned above, the new SolidWorks add-in is **NOT** compatible with Chrono 8.0

closed 2023-10-16 by rserban