projectchrono/chrono #470: ModuleNotFoundError: No module named 'pychrono' on M1 Mac

e-chesoni, 2023-11-06:
Hi, I'm running into issues running pychrono after following the [installation guide here](https://api.projectchrono.org/pychrono_installation.html). I have a simple python file (test.py) where I'm just trying to import pychrono to start
`import pychrono as chrono`
I get the following error
`ModuleNotFoundError: No module named 'pychrono'`
Things I've checked/verified:
`echo $PYTHONPATH` returns `.../opt/anaconda3/envs/chrono/share/chrono/python`
`conda list -n chrono` shows that pychrono is installed in the chrono env

Additional information
Chip: Apple M1 Pro
macOS: 13.5.2
Python Version in chrono: 3.9.18

I was not able to run the following lines as directed in the instructions:
conda install -c conda-forge mkl=2020
conda install -c conda-forge pythonocc-core=7.4.1
conda install -c nvidia/label/cuda-11.7.0 cuda-toolkit

for mkl, I was not able to find the package using the anaconda GUI. Is this compatible with macOS? Is there an alternative I should use?

for pythonocc, I installed the following version via the anaconda GUI:
pythonocc-core            7.7.2

for nvidia/label/cuda-11.7.0 cuda-toolkit I also was not able to find this in anaconda. Is this compatible with macOS? Is there an alternative I should use?

In [issue #431](https://github.com/projectchrono/chrono/issues/434) it seems TIE666 was able to get the import pychrono statement working with an M1 mac, so I have to believe I've installed something incorrectly or something. 

Can you please advise as to how I can get this project up and running? 

Happy to provide more info as needed, and very excited to use Pychrono!

rserban, 2023-11-06:
I have not tried installing the conda PyChrono in a while, but just looking on anaconda.org I can see that some of those packages are available:

- [mkl 2020](https://anaconda.org/conda-forge/pythonocc-core/files?page=2)
- [pythonocc-core 7.4.1](https://anaconda.org/conda-forge/pythonocc-core/files?page=3)

Not sure what your "anaconda GUI" finds or does not find, but maybe you should try to do this from the command line as listed in the PyChrono installation instructions.

Note that Chrono can ***only*** work with Cascade 7.4 (see comments [here](https://api.projectchrono.org/module_cascade_installation.html)), so installing a newer version (7.7.2) will not fly.

Finally, the MacOS PyChrono conda packages do not include any of the Chrono modules that rely on GPU computing through CUDA because Apple does not use NVIDIA GPUs anymore.  So, no need to worry about installing the cuda conda package.

e-chesoni, 2023-11-06:
Thanks for the response @rserban! Good to know CUDA is not needed.

I should mention, I tried installing mkl and pythonocc in the anaconda GUI (navigator?) because the command line method failed. As for mkl 2020 it looks like it's only available for "intel and compatible processors". 

[image]

When I use the navigator to search on a windows machine I do indeed find the mkl package, but not on an M1 mac. Does this mean pychrono is only compatible with intel macs?

Similarly, the commands `conda install -c conda-forge pythonocc-core=7.4.1` or `conda install pythonocc-core=7.4.1` did not work for me, so I tried using anaconda navigator to install. The only available version searching on an M1 mac is 7.7.2.

rserban, 2023-11-07:
I checked the scripts we use to generate the conda packages and it turns out that the Chrono::PardisoMKL module is not included in the MacOS PyChrono package. So MKL is also not needed.

OCC support is included though and, as I mentioned before, the Chrono::Cascade module is only compatible with an older OCC version. The lack of a corresponding pythonocc conda package is an issue.

To answer your question, because Apple does not use Intel processors anymore, we have recently decided to only generate and offer conda packages for ARM Macs.  Now, it looks like we need to reconsider the inclusion of Cascade support in the PyChrono conda package for Macs.

Having said that, I think the rest of PyChrono functionality should work anyway and yet it appears you cannot even load the core Chrono module. So there are likely additional issues here.  I do not have access to a Mac, so this is as far as I can help. Hopefully a Mac user can chime in and provide some additional insights.

e-chesoni, 2023-11-08:
Ok, good to know Chrono::PardisoMKL is not required for MacOS. I will try reinstalling pychrono with only the required packages in a new env and see how that works.

I'll keep checking here if any M1 Mac users have some advice.

Thanks for your help @rserban

e-chesoni, 2023-11-13:
Trying again on Mac;
I re-installed the pychrono in a new env following the instructions [here](https://api.projectchrono.org/pychrono_installation.html)
I was able to install all packages as directed with the exception of `conda install -c conda-forge pythonocc-core=7.4.1`. Had to install from source using the instructions [here](https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md). I only needed to install [swig](https://www.swig.org/) (which I did through Anaconda) to successfully install pythonocc-core version 7.4.1.
[image]
Finally, I installed Spyder through Anaconda, and tried running `demo_CH_coords.py`.  Again, I run into the import error:
```
runfile('/Users/elainechesoni/anaconda3/pkgs/osx-arm64_pychrono-8.0.0-py310_3631/share/chrono/python/pychrono/demos/core/demo_CH_coords.py', wdir='/Users/elainechesoni/anaconda3/pkgs/osx-arm64_pychrono-8.0.0-py310_3631/share/chrono/python/pychrono/demos/core')
First tutorial for PyChrono: vectors, matrices etc.
Traceback (most recent call last):

  File ~/anaconda3/envs/chrono/lib/python3.9/site-packages/spyder_kernels/py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File ~/anaconda3/pkgs/osx-arm64_pychrono-8.0.0-py310_3631/share/chrono/python/pychrono/demos/core/demo_CH_coords.py:17
    import pychrono as chrono

ModuleNotFoundError: No module named 'pychrono'
``` 
If anyone has resolved this issue with an M# mac, please let me know how it was done.

nalipour, 2025-10-22:
Hi @uebian !
It seems the conda command did not install the pychrono library.
Would you have any advice on how to install?
Looking forward to use pychrono ! Thanks for your help

uebian, 2025-10-25:
@nalipour We have built a new conda package (6429) for macOS, and the issue should have been addressed. Let me know if you still encounter issues.

closed 2025-11-04 by uebian