projectchrono/chrono #479: PyChrono ChFunction example is out of date

T-Lind, 2024-02-24:
Hi all,

I noticed that the following example:
https://github.com/projectchrono/chrono/blob/main/src/demos/python/core/demo_CH_functions.py
is outdated and runs with an error in its current state (the error stating that `ChFunction_Repeat` doesn't take any arguments.

Instead of passing `f_seq` into the constructor, it should be set as the following:
```python
f_rep_seq = chrono.ChFunction_Repeat()
f_rep_seq.Set_fa(f_seq)
```

dariomangoni, 2024-02-25:
Hi T-Lind,
thanks for pointing it out: ChFunctions are indeed one of the oldest part of the code and they have been one of the first to have been reviewed in our current (ongoing) refactoring of the code (see branch [overhaul](https://github.com/projectchrono/chrono/tree/feature/overhaul)).
We will take care of review the entire ChFunction page quite soon!

Dario

dariomangoni, 2024-06-20:
Hi @T-Lind, I think that the demo is now fixed as well as the documentation.
Please let us know if you see something else off!

closed 2024-06-20 by dariomangoni