projectchrono/chrono #486: Bug in FindThrust.cmake that breaks Chrono Multicore cmake build

CleConor, 2024-03-25:
In the actual version of Thrust (available together with CUDA Toolkit) in the version.h file, the define THRUST_VERSION has a comment in the same line and the regex that serves to bring out only the number doesn't take into account the comment.
Here is my fix:
```CMAKE
string( REGEX REPLACE
  "#define THRUST_VERSION[ \t]+([0-9x]+)(.*)"
  "\\1"
  version
  "${version}"
)
```

rserban, 2024-03-25:
What version of CUDA?

CleConor, 2024-03-25:
> What version of CUDA?

12.4

rserban, 2024-03-29:
I confirm this issue with CUDA 12.4.  I will push a commit with the fix you suggested (thanks!) and we will look for a better overall solution to finding Thrust (from a CUDA distribution or a stand-alone Thrust installation).

commit 0390d5a5 (2024-03-29): Fix issue with identifying Thrust version in latest CUDA release.

Note that this is a temporary fix that allows using Thrust from a CUDA 12.4 distribution. A better solution will be provided later.
Solution provided by CleConor
Closes #486.
closed 2024-03-29 by rserban