projectchrono/chrono #489: Dockerfile for uwsbel/packages:ubuntu

WYK96, 2024-04-03:
I'm learning how chrono is deployed on docker. If I understood correctly, the official docker image as described in https://hub.docker.com/r/uwsbel/projectchrono is built using the dockerfile in https://github.com/lliangthomas/sbel-docker/blob/lunar/Dockerfile. The https://github.com/lliangthomas/sbel-docker/blob/lunar/Dockerfile is built upon the layer uwsbel/packages:ubuntu. 

My question is: where can I get the dockerfile for uwsbel/packages:ubuntu? 

Thank you very much for the great work!

WYK96, 2024-04-03:
I've read the tech report [Integration of Docker Containers and Project Chrono](https://sbel.wisc.edu/wp-content/uploads/sites/569/2023/05/TR-2023-04.pdf). Maybe the advice from @lliangthomas would be very helpful :)

lliangthomas, 2024-04-05:
Hi,

The uwsbel/packages:ubuntu image was not only built through a Dockerfile (docker commit, etc.) so there is no Dockerfile for it. Let me know if you have any more questions

uebian, 2025-09-29:
Our building environment is now `uwsbel/packages:cuda12.8.1` and the docker file can be accessed here: https://gitlab.com/uwsbel/ubuntu-ci-env

closed 2025-09-29 by uebian