projectchrono/chrono #493: Bug? OpenGL does not work in combination with direct solvers

paulotto, 2024-04-19:
Hello,

there seems to be a bug if OpenGL is used together with a direct solver like Pardiso MKL.

The following line seems to be the issue causing the simulation to crash:

https://github.com/projectchrono/chrono/blob/b32083f677353d1d2b2e36ace6a4063750e2444d/src/chrono_opengl/UI/ChOpenGLStats.cpp#L305

Best,
Paul

rserban, 2024-04-19:
Paul - thanks for pointing this out.  Should be a simple fix. I'll look into it.
--Radu

paulotto, 2024-04-19:
You're welcome.

Best,
Paul

rserban, 2024-04-19:
Paul - I'll push a fix for this in a few minutes.  

Having said that, I was wondering why you use the Chrono::OpenGL run-time visualization in the first place.  Note that this module provides relatively limited capabilities as it is meant first and foremost for rendering results from large scale simulation (granular dynamics or SPH).  Hence the hardcoded assumption that one uses an iterative solver (left over from when this visualization module was working only with what is now Chrono::Multicore).

Anyway, everything should be working fine now, but, if possible, I recommend you use the Irrlicht, or even better the VSG options in Chrono.

closed 2024-04-19 by rserban

paulotto, 2024-04-20:
Thank you very much. I'll keep that in mind.

commit 6fcac488 (2024-04-19): Fix hardcoded assumption on current solver type in Chrono::OpenGL stats.

- render stats specific to a iterative VI or direct LS solver
- properly increment counter for calls to the Solve() function of a ChDirectLS solver
- update demo_GL_inv_pendulum to illustrate OpenGL stats rendering when using different solvers

Fixes #493