projectchrono/chrono #502: Get listed on the FMI tools page

chrbertsch, 2024-06-13:
I would like to encourage you get listed on the FMI tools page https://fmi-standard.org/tools/, see https://github.com/modelica/fmi-standard.org/blob/main/CONTRIBUTING.md#updating-the-tools-list.

dariomangoni, 2024-06-20:
Hi Christian, that's actually a nice idea!
We'd like to add some more features both to Chrono itself and to our [fmu_tools](https://github.com/DigitalDynamicsLab/fmu_tools) before getting listed there, but it's surely a thing that we will keep in mind!

chrbertsch, 2025-05-02:
Any news on this?

rserban, 2025-05-02:
Hi Christian,  I will do this soon.  We just submitted a paper on the Chrono FMI support we developed, so this is a good time to get these tools listed on the FMI tools page.  Thanks for the reminder.