projectchrono/chrono #506: cuda 12.5

cekees, 2024-07-09:
I saw during configuration that chrono does not support cuda 12.5 yet. Is that a hard constraint, or should I modify the cmakefile and try it?

rserban, 2024-07-10:
Unfortunately, it's not as simple as modifying the CMakeLists file.

There are several issues with using CUDA 12.5 in Chrono:
- the latest version of Thrust (2.4.0) included with the CUDA 12.5 toolkit does not allow thrust/ headers in host-only TUs when THRUST_DEVICE_SYSTEM=CUDA and only allows thrust/ headers in host-only TUs  when THRUST_DEVICE_SYSTEM={CPP, OMP, TBB}. 
This will require a refactoring of the internal multicore collision detection algorithms and of the Chrono::Multicore module, both of which rely on Thrust with the OpenMP backend.
- the API for cusparse and cublas has changed, with some functionality (e.g. incomplete decomposition preconditioners) completely removed.
This will require a refactoring of the Chrono::FSI module (and possibly the Chrono::GPU module).

All of the above will have to happen at some point, but I do not have the bandwidth to tackle any of it for quite a while.

cekees, 2024-07-10:
Thanks, @rserban. OK, it likes like I may be able top install CUDA 12.3 alongside CUDA 12.5.

salhus, 2024-10-12:
@cekees If you don't mind answering. Did it work with CUDA 12.3? The documentation suggests that the GPU and FSI module were tested with Pascal architecture with CUDA 11.4. 
I have Ada with 12.3, but I am having errors about a mismatch between the static and dynamic library configuration settings. If you know a configuration that works, it will be a massive help.

Cheers,
sal

Huzaifg, 2025-03-06:
The main branch of Chrono now supports CUDA versions uptil 12.8 on Windows and Linux (tested on Ubuntu 22.04)

closed 2025-03-06 by Huzaifg