projectchrono/chrono #511: Gravitational force by a planet/moon

wedesoft, 2024-08-17:
Hi, I would like to use ProjectChrono for a space simulator.
How can I implement a force which depends on the position of a spacecraft relative to a planet's position in order to simulate takeoff, orbiting and landing of a spacecraft?
Kind regards
Jan

wedesoft, 2024-08-17:
Also the position of the moon changes over time.

rserban, 2024-08-20:
The most flexible way to do this is probably to use [ChLoadBodyBody](https://api.projectchrono.org/classchrono_1_1_ch_load_body_body.html).

Derive your own class from this and implement the virtual method `ComputeBodyBodyForceTorque`. You can access the two bodies with `GetBodyA` and `GetBodyB` to get their current positions and orientations.  

See [ChLoadBodyBodyBushingGeneric](https://github.com/projectchrono/chrono/blob/6f6bb4c6103de2764f38e9f27076818d17ab1fd3/src/chrono/physics/ChLoadsBody.cpp#L516) for an example.

wedesoft, 2024-08-20:
Thanks for your response! Ok, will try. In the past I have successfully used Runge Kutta 4 integration to get stable orbits :)

wedesoft, 2024-08-20:
Awesome, it seems to work as long as the time steps are small enough (source code: [chronoorbit.cc](https://github.com/wedesoft/chronotest/blob/c5112d209902bdf092eb3427e8862e0a5a7b2f54/chronoorbit.cc)).
I had to compile it with option `-DEIGEN_MAX_ALIGN_BYTES=32` because it was crashing otherwise.
[image]

closed 2024-08-30 by rserban