projectchrono/chrono #515: conda install pychrono can not use cascade, Irrlicht, and other extensions

MRiabov, 2024-09-06:
Sup pychrono,
I'm a first time user, and `conda install pychrono` does not see pychrono.cascade, or pychrono.irrlicht, or any of the extensions.

You have the command here: https://projectchrono.org/pychrono/, which is the main page of the project (and where I expected to download it), but as I needed to import shapes it didn't work.
[image]

So point is - you probably should update the description for users like me that extensions won't work
Also mention it here: https://api.projectchrono.org/development/pychrono_installation.html

rserban, 2024-09-06:
@MRiabov  - please follow the installation instructions for the conda PyChrono package at: https://api.projectchrono.org/development/pychrono_installation.html.

As explained there, the command on the banner you posted will not install a proper version of PyChrono.

MRiabov, 2024-09-07:
Which is what I did already, but only 4-5 hours and questioning my sanity later after I first downloaded the package. You know.

closed 2025-11-26 by uebian