projectchrono/chrono #516: Create checkpoint file in pychrono

TheCrazyT, 2024-09-06:
So I guess this is not possible, since there is no way to use ChStreamOutAsciiFile etc. .
Are I'm correct?

TheCrazyT, 2024-09-06:
now i found [SerializeToJSON](https://github.com/projectchrono/chrono/blob/9.0.1/src/chrono_swig/interface/core/ChSystem.i#L4) but what is the opposite?

dariomangoni, 2024-09-08:
Hi @TheCrazyT,
the short answer is that **is not currently possible, except for the only method that you already found**.

As you may have seen from the [CHANGELOG](https://github.com/projectchrono/chrono/blob/9.0.1/CHANGELOG.md) there have been significant changes to the API. One of this, is obsoleting the `ChStreamOutAsciiFile` class and replace it with pure STL streams (of which the `ChStream[In|Out]AsciiFile` were basically slightly more than simple wrappers) that we didn't expose through SWIG yet.

However, this is not the reason for which the serialization/checkpointing is not fully implemented for the PyChrono version. What is truly missing are all the classes in the [chrono/serialization](https://github.com/projectchrono/chrono/blob/9.0.1/src/chrono/serialization) folder! These classes leverage some complex C++ feature, together with an heavy use of templates, thus being quite difficult to wrap in SWIG.

That's why the only thing that we exposed in Python and C# is _SerializeToJSON_ that allows us (especially for the Chrono::SolidWorks add-in) to:
- only export
- only the entire system
- only to JSON

We may add some additional methods to expand the flexibility of the serialization also for the wrapped languages, but we do not have strict plans for that!

rserban, 2024-09-08:
Could you please clarify: are you interested in serializing a Chrono system or in checkpointing? The two are *not* the same thing.

TheCrazyT, 2024-09-09:
Was interested in checkpointing.
But I aleady noticed that this feature is not even complete if you use a c++ program.

As a workaround I am currently using [dmtcp](https://github.com/dmtcp/dmtcp) for now.
Although I only use it with simple graph creation and without visualisation of the simulation.
Not shure if it would work for that, too.