projectchrono/chrono #517: OptiX link error

ceccocats, 2024-09-08:
branch: main (also tested on 9.0 release)
compile command:
```
cmake -DENABLE_MODULE_VEHICLE=ON -DENABLE_MODULE_OPENGL=ON -DENABLE_MODULE_IRRLICHT=ON -DENABLE_MODULE_POSTPROCESS=ON -DENABLE_MODULE_SENSOR=ON -DOptiX_INSTALL_DIR=/home/ceccocats/programs/NVIDIA-OptiX-SDK-7.7.0-linux64-x86_64/ -DENABLE_MODULE_ROS=ON -DENABLE_OPENMP=ON -DUSE_EIGEN_OPENMP=ON -DENABLE_MODULE_GPU=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=ON ..
```
I have Optix link error if i launch any sensors module demo.
This is the error with camera demo:
```
ceccocats@aron:~/programs/chrono/build$ ./bin/demo_SEN_camera 
Copyright (c) 2020 projectchrono.org
Chrono version: 9.0.1
Shader compile time: 1.93442
[ 2][    COMPILER]: COMPILE ERROR: 
Info: Pipeline statistics
        module(s)                            :     6
        entry function(s)                    :     6
        trace call(s)                        :     9
        continuation callable call(s)        :     0
        direct callable call(s)              :     0
        basic block(s) in entry functions    :   360
        instruction(s) in entry functions    :  6860
        non-entry function(s)                :     0
        basic block(s) in non-entry functions:     0
        instruction(s) in non-entry functions:     0
        debug information                    :    no
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Unresolved external symbol '_Z5powiffi' in module '__closesthit__material_shader'

[ 2][    COMPILER]: COMPILE ERROR: 
Info: Pipeline statistics
        module(s)                            :     6
        entry function(s)                    :     6
        trace call(s)                        :     9
        continuation callable call(s)        :     0
        direct callable call(s)              :     0
        basic block(s) in entry functions    :   360
        instruction(s) in entry functions    :  6860
        non-entry function(s)                :     0
        basic block(s) in non-entry functions:     0
        instruction(s) in non-entry functions:     0
        debug information                    :    no
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Unresolved external symbol '_Z5powiffi' in module '__closesthit__material_shader'

[ 2][    COMPILER]: COMPILE ERROR: 
Info: Pipeline statistics
        module(s)                            :     6
        entry function(s)                    :     6
        trace call(s)                        :     9
        continuation callable call(s)        :     0
        direct callable call(s)              :     0
        basic block(s) in entry functions    :   360
        instruction(s) in entry functions    :  6860
        non-entry function(s)                :     0
        basic block(s) in non-entry functions:     0
        instruction(s) in non-entry functions:     0
        debug information                    :    no
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Symbol '_optix_constant_memory_0x43991b1b6e28bbb2edd469bad5deca25' was defined multiple times. First seen in: '__miss__shader'
Error: Unresolved external symbol '_Z5powiffi' in module '__closesthit__material_shader'

```

ceccocats, 2024-09-08:
Solved!
The issue was a pow function call with arguments of mixed type (float, int):
https://github.com/projectchrono/chrono/blob/3eb56218b9fef0db58e5dd5b35a864e596bc2628/src/chrono_sensor/optix/shaders/material_shaders.cu#L1011
whit the passed arguments as float it works:
```
B_C = (1 + (1 - exp(-tan_ghalf_per_hC)) / tan_ghalf_per_hC) / (2 * pow(1.0f + tan_ghalf_per_hC, 2.0f));
``` 

I tried calling pow with integers and i always have compile errors.
Do you know if there is a method to build this files at compile time and not at runtime?

ceccocats, 2024-09-08:
Actually its possible to build PTX file at runtime with this flag `-DUSE_CUDA_NVRTC=OFF`.
With runtime compilation i found another issue:
At runtime nvdb_vol_intersect.cu is compiled without the flag `USE_SENSOR_NVDB` so it would not compile properly
https://github.com/projectchrono/chrono/blob/0704f62924ab5e09e1fdd2ffbbe9bf7a6d95c9f9/src/chrono_sensor/optix/shaders/nvdb_vol_intersect.cu#L9

rserban, 2025-11-26:
Fixed in commit #0704f629