projectchrono/chrono #518: Huge performance drop if colliders objects intersects

ceccocats, 2024-09-08:
branch: main
compile command:
```
cmake -DENABLE_MODULE_VEHICLE=ON -DENABLE_MODULE_OPENGL=ON -DENABLE_MODULE_IRRLICHT=ON -DENABLE_MODULE_POSTPROCESS=ON -DENABLE_MODULE_SENSOR=ON -DOptiX_INSTALL_DIR=/home/ceccocats/programs/NVIDIA-OptiX-SDK-7.7.0-linux64-x86_64/ -DENABLE_MODULE_ROS=ON -DENABLE_OPENMP=ON -DUSE_EIGEN_OPENMP=ON -DENABLE_MODULE_GPU=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=ON ..
```

I noticed that there is an huge performance drop in collision computation if objects intersects with each other.
Here is an example:
Take `demo_MBS_collision_trimesh` and modify the code to spawn fixed object more or less in the same position:
```
    for (int j = 0; j < 15; ++j) {
        auto falling = chrono_types::make_shared<ChBodyAuxRef>();

        // Set the COG coordinates to barycenter, without displacing the REF reference.
        // Make the COG frame a principal frame.
        falling->SetFrameCOMToRef(ChFrame<>(cog, principal_inertia_rot));

        // Set inertia
        falling->SetMass(mass * density);
        falling->SetInertiaXX(density * principal_I);
        falling->SetFixed(true);

        // Set the absolute position of the body:
        falling->SetFrameRefToAbs(
            ChFrame<>(ChVector3d(0, 0.3 + j*0.01, 0)));
        sys.Add(falling);

        falling->AddVisualModel(vis_model);
        falling->AddCollisionShape(coll_shape);
        falling->EnableCollision(true);
    }
```
This will induce a huge performance drop in the ComputeCollision stage.
This issue is annoying especially when you want to put static objects on an heightmap terrain that will eventually intersect with them.

Is this a limit of the current method or is possible to fix it?

ceccocats, 2024-09-08:
I tried on bullet3 (with a lot more cubes) and the performance is not affected with static object intersection.

rserban, 2024-09-08:
Place all these collision models in the same collision family (see [ChCollisionModel::SetFamily](https://api.projectchrono.org/classchrono_1_1_ch_collision_model.html#ab591335fd235fb5590e7204f4b7ad5bf)) and then disable collision between models in that family (see [ChCollisionModel::DissallowCollisionsWith](https://api.projectchrono.org/classchrono_1_1_ch_collision_model.html#a93ece48505bc18dce4fb53645f61b2bb)).

ceccocats, 2024-09-08:
Perfect, thank you!

closed 2024-09-08 by ceccocats