projectchrono/chrono #523: FEA demos much slower on v9 than v6

JasGarnier, 2024-10-06:
I downloaded the v6 pre-compiled package then I downloaded v9 source and compiled with visual C 2022. All fea demos are much slower in v9. I see that with the v6 visualisation demo the cpu is at 100% all the time and the gpu is at 12%. With v9 cpu usage fluctuates between 50 and 70% which gpu usage is 1 to 2%. I read a previous query where it was recommened to add the sys.SetNumThreads() command which I did but it made zero difference. What else might be the trouble?

rserban, 2024-10-14:
Things have changed significantly from version 6 to now. 
This also includes the various demos which now may use different settings and parameters, as well as different defaults for various Chrono settings.
It is quite likely that you end up comparing apples to oranges.
Did you check that you are using the same compiler and compiler options, problem definitions, number of threads, solver and integrator, solver and integrator settings, etc.?

JasGarnier, 2024-10-14:
Thanks for the reply. I am in the process of checking such things but without the original project files and source code it is quite challenging. Are they stored somewhere? The new fea examples do not work with the older libs or include files so I am having to detect the older names & locations for functions, objects etc. I tried the intel compiler with no change. OMP is clearly on but may be switched off by an IF block somewhere because the cpus should be at 100% and they aren't. I also notice blaze was used for V6 but not now. However, I had hoped someone else noticed this 2 to 3 times slowdown: The effort was already made to upgrade all fea examples so I cannot understand why a speed comparison wasn't also done. being as speed is vital for fea.

rserban, 2024-10-15:
Of course we monitor performance implications when we change code or add features.

You never mentioned which FEA demos you are looking at.  

There are several FEA formulations available in Chrono.  Corotational was also available in 6.0. Blaze was used in 6.0 and continues to be used only for the Chrono::Multicore module.

For FEA demos, Chrono does not use the GPU. Any GPU usage is due only to the run-time visualization. That usage may have changed because of changes in the frequency with which demos render a simulation.

Until you can confirm that you are comparing performance of the exact same physics problem, solved with the same (or equivalent) solver settings, we cannot draw any conclusions. 

You can get the source code for the 6.0.0 release by checking out the `release/6.0` branch.

JasGarnier, 2024-10-15:
Thanks that helped. I hadn't noticed the use_eigen_openmp switch in cmake before. It works fine now.

rserban, 2024-10-15:
Great!

By the way, ENABLE_OPENMP and USE_EIGEN_OPENMP should be enabled by default if your compiler supports OpenMP. Did you manually disable them?

closed 2024-10-20 by rserban