projectchrono/chrono #531: Some issues about Chrono::Solidworks

cyy227, 2024-11-25:
Hello! I hope this message finds you well. I have been using Chrono::SolidWorks recently for some projects, but I encountered an issue while running the spider robot demo file. Specifically, the error states that the module 'pychrono' has no attribute 'ChFunction_Sigma'. I checked your official website for this function but couldn't find any related information.

Additionally, I noticed other similar issues, such as 'chrono.ChFunctionSine' being incorrectly referred to as 'chrono.ChFunction_Sine'. I am using the latest V9 version and am unsure if this is a problem on my end or something more general. I would greatly appreciate it if you could provide any clarification or guidance to address these concerns.

dariomangoni, 2024-11-25:
Hi,
I assume you are talking about the python script: yes, we forgot to update it.
It's just a matter to update the commands to v9, following the CHANGELOG. We are going to do it soon.

Just please mind that the demo you are referring to it's not in this repository, but on https://github.com/projectchrono/chrono-solidworks

dariofusai93, 2024-11-25:
Fixed in Chrono::Solidworks commit 5847155bc8f22459e423ed657bb7ec01e4f56b93

closed 2024-11-25 by dariofusai93

cyy227, 2024-11-26:
Thank you both for the quick responses and helpful clarification. I truly appreciate your efforts in addressing this issue, especially the prompt fix in commit 5847155bc8f22459e423ed657bb7ec01e4f56b93.