projectchrono/chrono #532: Vehicle module can't find this part of vehiclecosim

wyj1111kkk, 2024-11-30:
Hi everyone! When I test template_project_vehicle_cosim inside chrono, it shows in cmake that the part vehiclecosim is not found. I have tried many times but still no solution, this problem has been bothering me for a long time, what is the reason for this?
[image]

rserban, 2024-11-30:
The reason is listed pretty clearly in the CMake output: your Chrono build was not configured with vehicle co-simulation support.
Go back to the Chrono CMake configuration and look carefully through the messages you get there.
The co-simulation support in Chrono::Vehicle **requires** MPI.  If an MPI installation is not found, then co-simulation support is disabled.
Once you get MPI installed, reconfigure and rebuild Chrono. Make sure you see that related demos are being built (they have names starting with `demo_VEH_Cosim_`.  After that, you should be able to correctly configure and build template_project_vehicle_cosim.

wyj1111kkk, 2024-12-01:
I see, thanks a lot, problem solved!

closed 2024-12-01 by wyj1111kkk