projectchrono/chrono #539: Missing CH_ROS_API for some header files

tengfoonglam, 2025-01-04:
Hi Project Chrono,

Was trying to compile the chrono_ros module and realised that some of the header files (which is not header-only) such as ChROSBodyHandler.h is missing the CH_ROS_API, which results in linker issues and the build to fail.

Would you like me to make a PR to fix this or will you amend this on your end? 

Thanks for the great work!

DanNegrut, 2025-01-04:
Yes, that would be very helpful and thank you.

Dan
---------------------------------------------
Bernard A. and Frances M. Weideman Professor
NVIDIA CUDA Fellow
Department of Mechanical Engineering
Department of Computer Science
University of Wisconsin - Madison
4150ME, 1513 University Avenue
Madison, WI 53706-1572
608 772 0914
http://sbel.wisc.edu/
http://projectchrono.org/
---------------------------------------------

AaronYoung5, 2025-01-22:
Hi! Just saw this, pushed a fix to the `feature/ros` branch (which I plan to merge into `main` shortly). 

Also, are you using Chrono::ROS on Windows? We don't actually build the ROS module on our windows CI, so if you are, I can look into adding this to one of our test cases.

closed 2025-11-26 by rserban