projectchrono/chrono #540: Invalid or corrupted .obj file

wyj1111kkk, 2025-01-06:
Sorry for the interruption. When I try to do the vehicle side of things, the .obj file does not open when the final compilation results in the following situation. But when I link that .obj file, Visual Studio 2022 reports an error showing invalid or damaged parts. What is the reason for this?
[image]

dariomangoni, 2025-01-07:
As you can see the specified path is a _relative_ path so the location when it expects to find the file depends on your current working folder.
Can you please go to your build directory and:
- check that you do have a _bin/data_ folder?
- try to run the demo by manually going into _bin/Release_ (or _bin/Debug_)

If this works then it's just a matter of a wrong working folder: did you manually changed the configuration of demos in Visual Studio?
In VS, try to right-click on your demo, Properties>Configuration Properties>Debugging and make sure that the Working Directory is set to _bin/Release_ (or _bin/Debug_).

wyj1111kkk, 2025-01-14:
I'm sorry, I'm a newbie. I would like to use the chrono project you developed to simulate the wheel-ground interaction. I used the demo_VEH_SCMTerrain_RigidTire.cpp in the demo in chrono and wrote a CMakeList. But I checked it as you said, and it still didn't solve the problem, and the .exe file generated by visual studio compilation shows that the .obj file can't be opened. Is the obj file to be linked in CMakeList? Below is the corresponding CMakeList.
“cmake_minimum_required(VERSION 3.18)
cmake_policy(SET CMP0091 NEW)
 Set the project name
project(my_project)
LIST(APPEND CMAKE_PREFIX_PATH "${CMAKE_INSTALL_PREFIX}/../Chrono/lib")
find_package(Chrono
             COMPONENTS Irrlicht Vehicle Postprocess
             OPTIONAL_COMPONENTS PardisoMKL
             CONFIG)
if (NOT Chrono_FOUND)
  message("Could not find Chrono or one of its required modules")
  return()
endif()
set(CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_STANDARD ${CHRONO_CXX_STANDARD})
if(APPLE)
    set(CMAKE_MACOSX_BUNDLE ON)
endif()
include_directories(${CHRONO_INCLUDE_DIRS})
if(${CMAKE_SYSTEM_NAME} MATCHES "Windows")
  if(MSVC AND ${MSVC_VERSION} GREATER_EQUAL 1915)
    add_definitions( "-D_ENABLE_EXTENDED_ALIGNED_STORAGE" )
  endif()
endif()
if(MSVC)
    add_definitions("-D_CRT_SECURE_NO_DEPRECATE")  # avoids deprecation warnings
    add_definitions("-D_SCL_SECURE_NO_DEPRECATE")  # avoids deprecation warnings
    add_definitions( "-DNOMINMAX" )                # do not use MSVC's min/max macros
endif()
add_executable(my_demo demo_VEH_SCMTerrain_RigidTire.cpp)
target_compile_definitions(my_demo PUBLIC "CHRONO_DATA_DIR=\"${CHRONO_DATA_DIR}\"") 
target_compile_options(my_demo PUBLIC ${CHRONO_CXX_FLAGS})
target_link_options(my_demo PUBLIC ${CHRONO_LINKER_FLAGS})
target_link_libraries(my_demo ${CHRONO_LIBRARIES})
add_DLL_copy_command()”

 

  
 
  
 In VS, try to right-click on your demo, Properties&gt;Configuration Properties&gt;Debugging and make sure that the Working Directory is set to bin/Release (or bin/Debug).

wyj1111kkk, 2025-01-14:
I'm sorry, I'm a newbie. I would like to use the chrono project you developed to simulate the wheel-ground interaction. I used the demo_VEH_SCMTerrain_RigidTire.cpp in the demo in chrono and wrote a CMakeList. But I checked it as you said, and it still didn't solve the problem, and the .exe file generated by visual studio compilation shows that the .obj file can't be opened. Is the obj file to be linked in CMakeList? Below is the corresponding CMakeList.
[image]
[image]

wyj1111kkk, 2025-01-16:
Followed what you said and my problem is solved . Thank you very much.

closed 2025-02-06 by wyj1111kkk