projectchrono/chrono #544: where can i find the blender-add-on which works with newer Blender versions?

luckychris, 2025-02-07:
I found this documentation

https://api.projectchrono.org/introduction_chrono_blender.html

but i am too dumb to find a running downloadable and installable add-on - file.

thx for any hint.

rserban, 2025-02-07:
The Python script for the add-on is in the Chrono distribution in [src/imporer_blender/](https://github.com/projectchrono/chrono/tree/main/src/importer_blender). The comments at the top of the file `chrono_import.py` explain how to add it to Blender.

Having said that, we should add those instructions to the documentation page.

tasora, 2025-02-07:
BTW I can add this: I just made a change to chrono_importer.py to make it compatible with latest Blender 4.3. Get this new version of the py add-in from the features/multidomain
branch, not the default master branch. The updated file will be merged to main branch later this month.
(The blender add-in in the master branch of git was working well up to Blender v.4.01, but fails from v.4.1 onward)

Alessandro

https://api.projectchrono.org/introduction_chrono_blender.html

luckychris, 2025-02-08:
thank you very much!

closed 2025-02-08 by luckychris