projectchrono/chrono #546: Change Git Branch Structure for Releases

jibril-b-coulibaly, 2025-02-12:
The `release/X.0` branches have redundant, un-merged and inconsistent information that I propose to streamline, or get rid of.

As tags and versions point to a specific point in history,  users can build any release version from the commit associated with a particular tag, e.g.,  b048d0b70897aa87a3bb39d089d7606f708d9a07 for 6.0.0. Release branches from `release/1.0` up to `release/6.0` do not have commits ahead of `main`, and tags from `1.0.0` up to `6.0.0` are associated with a specific commit that currently lives on the `main` branch: I would argue these branches could be deleted.

Given that, what is the rationale for keeping these branches? They seem to only provide an easier browsing to specific versions at best, and some are a source of issues (see below).

# Release 7.0 and above

Release branches `release/7.0`, `release/8.0`, and `release/9.0` have, in my opinion, a very confusing and possibly problematic organization.

These branches contain commits ahead of `main`, which seems to point at code existing on these `release` branches but not on `main`.

* First, this is confusing because some of these commits are `cherry-pick` from `main`, creating duplicates that are difficult to track.
For example, on branch `release/9.0`, the first 23 out of the 27 commits have been cherry-picked from `main` and branching could have been avoided there.

* Second, some commits on the release branches have either:
  - not been merged into `main`.
  - been cherry-picked back into `main` later

  This breaks the expected timeline, that `main` should contain all the code from a prior release + newer development code.
For example, on branch `release/9.0`, commit 312254b51f0e69c4be842632a88750c70479ea85 fixes an incorrect declaration that is not present on `main` as of [today](https://github.com/projectchrono/chrono/blob/988297616c4b7ec77906abe28620baba5f62b1bb/src/chrono/physics/ChHydraulicActuator.h#L73), and commit 2617649bf687456328a122b63dc0bb64df91394a was cherry-picked back into `main` (b45b1055b5e0068b3dd254fdd519426f4fb6ea0f) a day later.

### Summary

1. some branches that are not in active development (`release/1.0` up to `release/6.0`) simply point at a commit in the history. This might be useful but is not necessary
2. some tags reside with commit that are not on `main` but instead on `release` branches that are partially copied from `main` and partially unmerged into `main`, breaking the expectation that any code on `main` that post-dates a given tag should have all the code from that tag, plus, newer development code.

# Proposed change

## Immediate changes

* Merge all development on release branches `release/7.0`, `release/8.0`, and `release/9.0` into `main` so that no "release" code that exists in these older tags is left out of the version-of-the-day `main`.
  * Although there is no way to make this retroactive without breaking the history, I believe it is developers' expectation that `main` contains all previous code, especially release code. After merging this outstanding release code, this expectation will at least be satisfied between now and the next release.
  * This should be properly documented in future release notes

## Future workflow change

* Create a Release and associated Tag directly in `main`
  * Creating release branches is not necessary for tagging a particular release
* If desired, create release branch at the corresponding tag commit for convenience
  * As for `release/1.0` up to `release/6.0`, I can understand that it can make browsing easier to have these branches
  * If not for convenience, I don't see a reason why we have one branch per release: we could simply use the tags to go to older versions, and have a single `release` branch that tracks the latest release (this is what is done in [LAMMPS](https://github.com/lammps/lammps) for example). 

I will be happy to help make these changes, or make them myself.

Jibril

rserban, 2025-02-12:
Thank for the detailed comments.
We used to use GitFlow for releases as well, but decided to switch to this new model because it makes it easier to deal with patches to previous releases (and this was important for our own work and for some of our external partners).  This was an internal decision based on our own workflow and release timelines.
Users should just stick with using the main branch or one of the [releases](url).
(commit [312254b](https://github.com/projectchrono/chrono/commit/312254b51f0e69c4be842632a88750c70479ea85) was a mistake; the code in the main branch is correct).

closed 2025-02-12 by rserban

jibril-b-coulibaly, 2025-02-12:
Thank you for your prompt answer and for the historical perspective from a long time developer.

Am I correct to understand that branches for release 7 and over are maintained and patched (e.g., by cherry-picking bug fixes commits from main) for specific users/collaborators using these release versions?

Regarding workflows, there are still several feature branches. For developing new modules are you recommending using trunk-based branches from main ?
Thank you