projectchrono/chrono #547: the visualization interface is not responding

wyj1111kkk, 2025-02-17:
Hi, when I try to run the demo_VEH_SCMTerrain_FEATire example, the visualization interface is not responding, is this due to my computer's lack of performance or something else?Do I need to increase the performance of my computer to solve this problem?

[image]

rserban, 2025-02-17:
That demo is indeed quite slow (because of the FEA calculations). On my machine, it runs with a real-time factor of 1000 (that is, 1000 times slower than your wall clock).  You can check what that value is on your machine: with the focus on the visualization window, press the "i" key and you will get a panel with simulation information, including the current RTF.

rserban, 2025-02-18:
Did you build Chrono in Release?

wyj1111kkk, 2025-02-18:
Yes, is there a difference between debug and release?

rserban, 2025-02-18:
It's more than just "a little more efficient"; there is a significant difference in performance between Release and Debug builds.
This is most likely due to your particular hardware. On my laptop (with a newer and better processor), I get an RTF of about 650.

One thing you can try (other than finding a better machine) is to set the number of OpenMP threads which can help some both the FEA and SCM calculations. Assuming you have at least 4 physical cores, try adding
`sys.SetNumThreads(4, 1, 1); `

closed 2025-09-09 by rserban