projectchrono/chrono #550: Arithmetic optimizations

jibril-b-coulibaly, 2025-02-23:
Hello,

I am developing a new module and taking inspiration from the existing code.
I notice many instances of floating point operations that I find unusual in terms of readability, code duplication, or performance, e.g.,

1. square roots using `pow(x, 0.5)` and cubic root using `pow(x, 1. / 3.)` instead of `sqrt(x)` and `cbrt(x)`
https://github.com/projectchrono/chrono/blob/09caa3e02cccf0a7d571264026ff3dbf430c35d2/src/chrono_modal/ChModalAssembly.cpp#L710

2. powers using `pow(x, N)` instead of `x*x*...`
https://github.com/projectchrono/chrono/blob/09caa3e02cccf0a7d571264026ff3dbf430c35d2/src/chrono/fea/ChBeamSectionCosserat.cpp#L154-L158

3. divisions
https://github.com/projectchrono/chrono/blob/09caa3e02cccf0a7d571264026ff3dbf430c35d2/src/chrono/core/ChTensors.h#L186

- For 1, I proposed #549

- For 2. although many modern compilers are usually able to eliminate repeated sub-expressions, or optimize `std::pow` for integer exponents, I found GCC to be much slower for an operation like `std::pow(x, 3)` compared to `x*x*x`, even with flag `-O3` (e.g., my Ubuntu Release build of Chrono with GCC 13.3.0, and tests on godbolt I could share if there is interest. Only tests with flag `-ffast-math` , which allows non-compliant code, showed negligible difference).
  - What are the motivations behind the heavy use of `pow` throughout the code base? Have benchmarks ruled out possible performance hits compared to bare multiplications?

- For 3. compilers likely won't replace the division by a reciprocal and faster multiplication for values that are not powers of 2 (for the same standard-compliant reasons discussed above). 
  - Is there a desire to replace expressions such as `x / 3.` by multiplications with constants that can be evaluated at compile time, e.g., `x * (1. / 3.)` (I saw plenty of those throughout the code) ? If so, would defining such constant expressions (e.g. `CH_1_3`, similar to e.g. `CH_PI` in `ChConstants.h`) be helpful for readability, and systematic replacement of `x / 3.` by `x * CH_1_3` preferred?

I appreciate your perspective and insight.
I would be more than happy to take on these tasks.

Jibril

rserban, 2025-02-24:
Jibril, a couple of comments:

1. using sqrt and cbrt where applicable is certainly preferable, both for performance and readability.
2. yes, those could/should be replaced (especially for squaring). For higher powers, it's a question of readability and preference. I personally tend to stay away from using pow, unless in some initialization function that is called once or very seldomly.  Some of us are more used to using pow and/or don't immediately think about performance (of course, there's also the danger of premature optimization).
3. this also could be implemented, but:
   - it could quickly get out of hand if we decide to do this for a ton of cases (1/3 is probably common enough to justify it)
   - we have code that can be configured in single or double precision; that may require two constants for each value.

I hope this makes sense. If so, it'd be great if you added some of 2 and 3 to your PR.  One issue with this kind of changes is that we always have many balls in the air in various feature branches (some of them quite long-lived), so there's always the issue of merge conflicts. But that's something we can deal with.

rserban, 2025-02-24:
On a different note: you mention working on a new module.  It'd be great if you could give us a bit more details (we can take this offline if needed).  The reasons I ask are:
- we should make sure this is not something someone is already working on or something we plan on doing.
- with any new module, there's the question of long-term maintenance and support.

closed 2025-03-25 by jibril-b-coulibaly