projectchrono/chrono #563: Bug in ChCollisionSystemBullet::RayHit(...) when using it to find collisions with just one specific model

bmo-Scape, 2025-04-08:
Using version 9.0.1.

Calling ```ChCollisionSystem::RayHit(const ChVector3d& from, const ChVector3d& to, ChCollisionModel* model, ChRayhitResult& result)``` when using bullet for the collision system leads the the code below being hit. But it doesn't perform the comparison with the desired model correctly.

The code finds all collisions with all models and then afterwards (in lines 386 to 392) finds the collisions that fit the desired model:
[image]

But this filtering is using the wrong model pointer. Instead of using ```rayCallback.m_collisionObject``` in line 387, ```rayCallback.m_collisionObject[i]``` should have been used. Otherwise, only if the last colliding object was the correct object, the correct collisions will be detected.

I hope that someone is able to take this info and turn it into a proper fix (PR) with the corresponding test that the code test then works correctly.

rserban, 2025-04-18:
@bmo-Scape - thanks for pointing this out. I will take a look and provide a fix when I get a chance.

rserban, 2025-11-26:
Sorry this took a while. I will push the fix in a minute

closed 2025-11-26 by rserban