projectchrono/chrono #576: optix: cmake configuration error

comejv, 2025-05-23:
Hello, I am trying to configure the main branch. cmake couldn't find optix so I supplied optix_root_dir as src/chrono_sensors/optix, but I still have these errors that won't let me finish the configuration:

CMake Error at cmake/FindOptiX.cmake:68 (file):
   file failed to open for reading (No such file or directory):

     .../chrono-src/src/chrono_sensor/optix/include/optix.h
 Call Stack (most recent call first):
   src/chrono_sensor/CMakeLists.txt:370 (find_package)

 CMake Error at cmake/FindOptiX.cmake:69 (string):
   string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
   command.
 Call Stack (most recent call first):
   src/chrono_sensor/CMakeLists.txt:370 (find_package)

 OptiX include directory: .../hrono-src/src/chrono_sensor/optix/include

Any idea?
Thanks.

dariomangoni, 2025-05-23:
Hi,
when we ask for a third-party library is because it's **not** included in the Chrono source.
Otherwise we would have pointed directly to our sources.
The OptiX that you need to provide is the one provided by NVIDIA and that you should install before linking to it.
It should be something like: "C:/ProgramData/NVIDIA Corporation/OptiX SDK 7.7.0"

comejv, 2025-05-23:
Awesome sorry for the confusion!

closed 2025-05-23 by comejv