projectchrono/chrono #581: Cone collisions in version 8

Oblomov, 2025-06-21:
Hello all,

in GPUSPH we offer the user the possibility to handle floating object motion and collitions through Project Chrono. We have recently revisited this integration, fixing a number of issues we had on our side in the way we handled the Chrono object setup starting from GPUSPH geometries, but we have been unable to set up a reliable correspondent for the GPUSPH cone.

(One note before I get deeper into the issue: we currently support versions 5 to 8 of Chrono —we do plan to support version 9, but the API changes are extensive and there is no simple way to support both 9+ and 8- at the moment, so this issue refers to version 8. I understand that it may be considered “legacy” so feel free to just close this issue as WONTFIX.)

There are three primary obstacles to realizing a GPUSPH cone in Chrono.

----

### not supported in the Bullet collision model

`AddCone()` simply returns `false`.

### incorrect bounding box

Due to a trivial copy-paste error, the bounding box for the cone is incorrect, since `cmin` is set twice and `cmax` is not set; this is trivial to fix:
```patch
diff --git a/src/chrono/geometry/ChCone.cpp b/src/chrono/geometry/ChCone.cpp
index 7191b07c5..b47d8ff95 100644
--- a/src/chrono/geometry/ChCone.cpp
+++ b/src/chrono/geometry/ChCone.cpp
@@ -28,7 +28,7 @@ ChCone::ChCone(const ChCone& source) {
 }
 void ChCone::GetBoundingBox(ChVector<>& cmin, ChVector<>& cmax, const ChMatrix33<>& rot) const {
     cmin = ChVector<>(-rad.x(), center.y() - rad.y() / 3.0, -rad.z());
-    cmin = ChVector<>(+rad.x(), center.y() + 2 * rad.y() / 3.0, +rad.z());
+    cmax = ChVector<>(+rad.x(), center.y() + 2 * rad.y() / 3.0, +rad.z());
 }
 
 void ChCone::ArchiveOUT(ChArchiveOut& marchive) {
```

### rotations not supported

This is an issue for us because GPUSPH uses the z axis as the vertical axis by convention, and the unrotated cone is z-aligned, while in Chrono it is y-aligned. Since the Chrono cone does not support rotation, even a static (unrotated) GPUSPH cone cannot be mapped to its Chrono equivalent.

----

As a “bonus”, we also have the issue that GPUSPH supports truncated cones, but AFAICS this is not supported by Chrono presently  (AFAIC, not in v9 either). This is however a secondary issue and should probably be addressed as a separate feature request.

### workaround: convex hull

I am currently exploring as a workaround the use of a point cloud assembled from the GPUSPH border filling. From preliminary tests this seems to be working.

rserban, 2025-06-25:
Happy to hear things are working @Oblomov .  

Indeed, it is not surprising to see differences due to the different collision shape (analytical cone vs. convex hull).

I opted to center the cone at its midpoint to be consistent with the Bullet definition of their cone collision shape.  Arguably, the apex or the base center would be better origins for such a shape.  The goal is for a user to easily place a shape within a model; from that point of view, the midpoint is only slightly less convenient than one those other two options. I personally think that the centroid is the worst choice of all; indeed, except for the trivial case where the collision model consists of only one cone and the body is assumed homogeneous, placing a cone based on its centroid is more cumbersome than any of the other 3 choices.

Relative translation and rotation of cone collision shapes in a collision model should be working fine now.  
By the way, in the latest Chrono code (9.0 and newer), the collision models were refactored and are now consistent with the visual models (in terms of API and underlying implementation).

Unless there are any other changes you need for Chrono 8.0, I'll go ahead and tag this as 8.1.0 and you will be able to target that for use with the current version of GPUSPH.

rserban, 2025-06-28:
@Oblomov  - I fixed the version and retagged. Should be all good now, but please let me know if you see any other issue

Oblomov, 2025-06-29:
Thanks, that seems to have done it.