projectchrono/chrono #599: _Replace With a Descriptive Title_

kjgbrt, 2025-08-06:
**Summary**
On Project Chrono webpage, there are two confusingly similar but different installation guides, I'd like clarification and removal of one of them.
**Chrono Version and Platform**
No version/platform, website specific

**Details**
Hello, please see the following two sites: 
1. https://api.projectchrono.org/development/install_guides.html
   * "Generated on Tue Aug 5 2025" at bottom of page
   * lists 19 modules
2. https://api.projectchrono.org/development/tutorial_table_of_content_install.html 
   * "Generated on Fri Dec 22 2023 06:37:43" at bottom of page
   * lists 20 modules (additional is exactly the module I'm interested in, OPENGL)

Are these sites in conflict with each other? Which one is considered up to date? I'd appreciate removing/deprecating one of the two. Thanks in advance.

kjgbrt, 2025-08-06:
Thanks for the quick response. I'm sorry to hear OpenGL is no longer supported, but good that I found out early. 
The dead link is directly on branch "main" page: https://github.com/projectchrono/chrono under "Build and install instructions"

Was support already dropped in v9.0?

rserban, 2025-08-06:
The old Chrono::OpenGL run-time visualization module was very limited and was provided only as a fast way of visualizing large-scale systems coming from granular dynamics simulations with the Chrono::Multicore module.  The newer Chrono::VSG module provides a lot more functionality with the same or better rendering performance.

Thanks for pointing out the link to the obsolete installation guides. I will fix that shortly.

kjgbrt, 2025-08-08:
Hi @rserban , to save a little bit of effort in making a separate but related issue, I also wanted to make a note specifically of Chrono v9.0: the build script and installation instructions for installing VSG in v9.0 has one very small but time consuming error. page: https://api.projectchrono.org/9.0.0/module_vsg_installation.html

issue: vsgImGui for Chrono v9.0 with VSG should NOT use "latest" version, it should use v0.6.0. 
solution: update page and build script contrib/build-scripts/vsg/buildVSG.bat line 64

is:
`     git clone "https://github.com/vsg-dev/vsgImGui" "download_vsg/vsgImGui"`
should be: 
`    git clone -c advice.detachedHead=false --depth 1 --branch v0.6.0 "https://github.com/vsg-dev/vsgImGui" "download_vsg/vsgImGui"`

a note in case someone in the future is lost and debugging like i have been: this will fix the error Unknown CMake command vsg_setup_dir_vars or similar

rserban, 2025-08-08:
Thanks for bringing this up.
The scripts in the current development code have already been modified (but the documentation is still incorrect).
But we will switch to using newer versions of the VSG libraries soon and will make sure the scripts and documentation agree.

closed 2025-09-09 by rserban