projectchrono/chrono #606: cmake Thrust configuration errors

dsfenn, 2025-09-04:
**Summary**

I am running into several errors regarding Thrust when running cmake. It looks like Thrust was installed with CUDA, as `/usr/local/cuda/lib64/cmake/thrust` exists.

**Chrono Version and Platform**

Chrono version is the latest `main`: 432a73a9b93522eaf2c23b8a5f554ebe49e0ded3
RHEL 8
cmake version 3.26.5
gcc 8.5.0
CUDA 13.0

**Expected Behavior**

Executing the included `buildChrono.sh` script (specifically, with `-DCH_ENABLE_MODULE_FSI:BOOL=ON` should build Chrono. 

**Actual Behavior**

Running `buildChrono.sh` gives the following output:

```
-- Searching for Thrust
--   Thrust not found. Set Thrust_ROOT to the location of Thrust library.
```

and 

```
CMake Error at src/chrono_fsi/sph/CMakeLists.txt:196 (target_link_libraries):
  Target "Chrono_fsisph" links to:

    Thrust::Thrust

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.
```

If I modify line 358 of `src/CMakeLists.txt` by changing it from

`find_package(Thrust)` to `find_package(Thrust REQUIRED CONFIG)`, Thrust is found, but then I receive the error

```
CMake Error: install(EXPORT "ChronoTargets" ...) includes target "Chrono_fsisph" which requires target "_Thrust_Thrust" that is not in any export set.
CMake Error in src/CMakeLists.txt:
  export called with target "Chrono_fsisph" which requires target
  "_Thrust_Thrust" that is not in any export set.
```

I'm not sure what to do about this error.

**Steps to Reproduce**

Copy `buildChrono.sh` to a directory outside the source tree. Modify the source, build, and Eigen directories accordingly. Configure it with the following options, then execute:

```
cmake -G "Unix Makefiles" -B ${BUILD_DIR} -S ${SOURCE_DIR} \
      -DCMAKE_INSTALL_PREFIX:PATH=${INSTALL_DIR} \
      -DCH_ENABLE_MODULE_IRRLICHT:BOOL=OFF \
      -DCH_ENABLE_MODULE_VSG:BOOL=OFF \
      -DCH_ENABLE_MODULE_OPENGL:BOOL=ON \
      -DCH_ENABLE_MODULE_VEHICLE:BOOL=OFF \
      -DIOMP5_LIBRARY=${IOMP5_DIR} \
      -DCH_ENABLE_MODULE_POSTPROCESS:BOOL=ON \
      -DCH_ENABLE_MODULE_MULTICORE:BOOL=ON \
      -DCH_ENABLE_MODULE_FSI:BOOL=ON \
      -DCH_ENABLE_MODULE_GPU:BOOL=ON \
      -DCH_ENABLE_MODULE_PARDISO_MKL:BOOL=OFF \
      -DCH_ENABLE_MODULE_CASCADE:BOOL=OFF \
      -DCH_ENABLE_MODULE_COSIMULATION:BOOL=OFF \
      -DCH_ENABLE_MODULE_SENSOR:BOOL=OFF \
      -DCH_ENABLE_MODULE_MODAL:BOOL=ON \
      -DCH_ENABLE_MODULE_MATLAB:BOOL=OFF \
      -DCH_ENABLE_MODULE_CSHARP:BOOL=OFF \
      -DCH_ENABLE_MODULE_PYTHON:BOOL=OFF \
      -DCH_ENABLE_MODULE_SYNCHRONO:BOOL=OFF \
      -DBUILD_BENCHMARKING:BOOL=ON \
      -DBUILD_TESTING:BOOL=ON \
      -DCH_ENABLE_OPENCRG:BOOL=ON \
      -DCH_USE_CUDA_NVRTC:BOOL=OFF \
      -DCH_USE_FAST_DDS:BOOL=ON \
      -DEIGEN3_INCLUDE_DIR:PATH=${EIGEN3_INSTALL_DIR} \
      -DIrrlicht_ROOT:PATH=${IRRLICHT_ROOT} \
      -DBlaze_ROOT:PATH=${BLAZE_ROOT} \
      -DOptiX_INSTALL_DIR:PATH=${OPTIX_INSTALL_DIR} \
      # -FastDDS_ROOT:PATH=${FASTDDS_INSTALL_DIR} \
      -DGLEW_DIR=${GL_INSTALL_DIR}/${LIB_DIR}/cmake/glew \
      -Dglfw3_DIR=${GL_INSTALL_DIR}/${LIB_DIR}/cmake/glfw3 \
      -DGLM_INCLUDE_DIR:PATH=${GL_INSTALL_DIR}/include \
      -DOpenCRG_INCLUDE_DIR:PATH=${CRG_INCLUDE_DIR} \
      -DOpenCRG_LIBRARY:FILEPATH=${CRG_LIBRARY} \
      -DOpenCASCADE_DIR:PATH=${CASCADE_INSTALL_DIR}/${LIB_DIR}/cmake/opencascade \
      -DSpectra_INCLUDE_DIR:PATH=${SPECTRA_INCLUDE_DIR}/include \
      -DMATLAB_SDK_ROOT:PATH=${MATLAB_INSTALL_DIR}/extern \
      -Dvsg_DIR:PATH=${VSG_INSTALL_DIR}/${LIB_DIR}/cmake/vsg \
      -DvsgImGui_DIR:PATH=${VSG_INSTALL_DIR}/${LIB_DIR}/cmake/vsgImGui \
      -DvsgXchange_DIR:PATH=${VSG_INSTALL_DIR}/${LIB_DIR}/cmake/vsgXchange \
      -Durdfdom_DIR:PATH=${URDF_INSTALL_DIR}/lib/urdfdom/cmake \
      -Durdfdom_headers_DIR:PATH=${URDF_INSTALL_DIR}/lib/urdfdom_heders/cmake \
      -Dconsole_bridge_DIR:PATH=${URDF_INSTALL_DIR}/lib/console_bridge/cmake \
      -Dtinyxml2_DIR:PATH=${URDF_INSTALL_DIR}/CMake \
      -DSWIG_EXECUTABLE:FILEPATH=${SWIG_EXE} \
      -DCMAKE_BUILD_TYPE="Release" \
      --log-level=VERBOSE
```

**Further Information, Files, and Links**

rserban, 2025-09-09:
I have a slightly different setup on my machine (I use different versions of CUDA through environment modules) and as such cannot reproduce this.

Having said that, I believe I know how this can be fixed.  Could you please try the following:
- confirm that the directory `usr/local/cuda/include/cccl/thrust` exists; if yes, then
- edit the file cmake/FindThrust.cmake (in the Chrono [source tree](http://github.com/projectchrono/chrono/blob/main/cmake/FindThrust.cmake)) and
- add `/usr/local/cuda/include/cccl`  in the list of HINTS to `find_path(THRUST_INCLUDE_DIR ...` (e.g., after line 38)

Let me know if this fixes the problem and I will push the "fix".

dsfenn, 2025-09-09:
Yes, that fixed the issue. Thank you!

commit 6417b4c3 (2025-09-15): Fix typo in CUDA 13 include directory

Fixes #606