projectchrono/chrono #615: issue with installation Pychrono 9.0.1

siddhu2310, 2025-09-30:
I'm trying to install pychrono 9.0.1, it required a lot of dependencies and getting error, Could you please update installation method or provide the yml file.

Also, which Python Version to use for proper installation with conda.

uebian, 2025-10-03:
Thank you for raising the issue. Our conda package building pipeline has recently changed and we haven't updated the document. We have simplified the installation process so you won't need to manually download and install packages as before. Could you please try using the following two command in the NEW environment to install PyChrono through conda? Thank you very much!

```
conda create -n chrono_test python=3.12 -c conda-forge
conda install pychrono -c projectchrono -c dlr-sc -c conda-forge
```

siddhu2310, 2025-10-03:
Thank you, it works